google/coding-with-chrome #11: Building fails on pristine Mac ElCapitan

MartenvanWezel, 2016-01-18:
1/ install git
2/ install node.js
3/ 
Last login: Mon Jan 18 00:44:34 on console
Martens-Mac:~ marten$ git clone git://github.com/google/coding-with-chrome.git
xcode-select: note: no developer tools were found at '/Applications/Xcode.app', requesting install. Choose an option in the dialog to download the command line developer tools.
Martens-Mac:~ marten$ git clone git://github.com/google/coding-with-chrome.git
Cloning into 'coding-with-chrome'...
remote: Counting objects: 1140, done.
remote: Compressing objects: 100% (249/249), done.
remote: Total 1140 (delta 107), reused 0 (delta 0), pack-reused 878
Receiving objects: 100% (1140/1140), 993.41 KiB | 335.00 KiB/s, done.
Resolving deltas: 100% (578/578), done.
Checking connectivity... done.
Martens-Mac:~ marten$ cd coding-with-chrome/
Martens-Mac:coding-with-chrome marten$ ls
CONTRIBUTING.md		build			resources
LICENSE-HEADER		doc			spec
LICENSE.md		documentation.json	src
README.md		karma.conf.js		test
app			package.json		third_party
Martens-Mac:coding-with-chrome marten$ git checkout jquery-turtle
Branch jquery-turtle set up to track remote branch jquery-turtle from origin.
Switched to a new branch 'jquery-turtle'
Martens-Mac:coding-with-chrome marten$ 
Martens-Mac:coding-with-chrome marten$ npm install
loadDep:jquery-turtle → n ▀ ╢██████████░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░╟
loadDep:jquery-turtle → n ▄ ╢██████████░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░╟
^C
Martens-Mac:coding-with-chrome marten$ 

(stuck)

MartenvanWezel, 2016-01-18:
(note: this was a brand new instance (vmware) of el capitan. Nothing installed except OSX updates.

MartenvanWezel, 2016-01-18:
Update: I am suspecting that this is actually an artefact of npm handling mac terminals poorly and a stuck network request.
 
The pretty progressbar hid all the actual messages, and I think one message was something like 'downloading <blah>....' so perhaps at some point it would timeout?

One thing that clearly is needed (and therefore we need to mention it in our readme) is java - I believe :

[image]

I also got a few arduino errors:
[14%] cwc.framework.Ev3 [Errors!] (1.8 sec)[Closure Compiler Error] 2 errors for genfiles/frameworks/internal/arduino_framework.js:ERROR -  read error.
Sample usage:  --compilation_level (-O) VAL --externs VAL --js VAL --js_output_file VAL --warning_level (-W) [QUIET | DEFAULT | VERBOSE]
Run with --help for all options and details

MarkusBordihn, 2016-01-18:
The reported issues are not related to "Coding with Chrome" itself and more npm related.
Could you please post your npm version and the npm log file.

Java JDK is not really needed and will be installed automatically by NPM if needed.

closed 2016-01-18 by MarkusBordihn

MartenvanWezel, 2016-01-18:
I'm sure you're right it's not CWC's fault, but I think it might be best to tell people explicitly to install JDK and the dev tools (xcode commandline stuff) because if you do, building CWC works on OSX.. I tried letting NPM download and install needed stuff but it kinda fails, and get stuck with a spinny ascii status thing:

[image]

If I install jdk-8u66-macosx-x64.dmg everything works.

MarkusBordihn, 2016-01-18:
I will add this to the readme.