google/coding-with-chrome #17: Demo Video

MartenvanWezel, 2016-01-19:
URL as discussed, image attached. Text, perhaps "Quick Intro"

[image]

Also put all video links in 'Resources' as well..

MartenvanWezel, 2016-01-19:
ps of course image should be scaled down to 'reasonable' :) don't need to dominate the main screen, but perhaps 1/5 of full screen size.

MarkusBordihn, 2016-01-19:
@MartenvanWezel 
Could you please provide the URLs, because I have left my laptop at work. ^^°

MartenvanWezel, 2016-01-19:
https://www.youtube.com/watch?v=Cbpug5Atmmo
https://www.youtube.com/watch?v=S3DhDiw0pXs
https://www.youtube.com/watch?v=R4XE3GW1EUc
https://www.youtube.com/watch?v=CkoJWKWIWFU

closed 2016-01-19 by MarkusBordihn

MartenvanWezel, 2016-01-19:
Hey Markus, can you also put the #1 video in the main CWC screen, with as 'icon' the image attached to this issue?

quick mockup:
[image]

MarkusBordihn, 2016-01-19:
I don't see any video which looked like the 'icon' so if you could provide me the video I could implement it.

I don't want to use one of the former videos because they very good but more for advanced users, so we should show an simple example with Blocks as quick intro instead.

MartenvanWezel, 2016-01-19:
True, we currently don't have a better video. :) - but video #1 does show a 'demo' of how to use CwC so I figured we would use that video until we build a better one. I hope Timothy will make one, else maybe I will

MartenvanWezel, 2016-01-19:
(I just pieced together the icon to show CwC in one of its most powerful modes, not intended to be a representation of the actual video just yet)