google/coding-with-chrome #19: Windows build error

carlosperate, 2016-01-24:
Hi,

I had a play with coding-with-chrome at the BETT show in London, and it is quite awesome, great work guys!

I tried installing this locally (Windows 10 x64, node 5.0 x64) and encountered the following error:

```
> coding-with-chrome@2.1.21 build-cwc-files E:\workspaces\git\coding-with-chrome
> node build/cwc_files.js

[14%] cwc.ui.Builder Compiling 52 soy files to C:\Users\Carlos\AppData\Local\Temp\closure-builder-9H34MN4 (0.0 sec)soynode: Compile error
 Error: spawn java ENOENT
[Soy Compiler Error] Failed for C:\Users\Carlos\AppData\Local\Temp\closure-builder-9H34MN4:Error: Error compiling templates
```

Then if I tried to run the chrome app anyway, it gets locked at "Building the Coding with Chrome UI" build step, which I guess it would be expected.

carlosperate, 2016-01-24:
Tried again with the [tagged version 2](https://github.com/google/coding-with-chrome/releases/tag/v2.0.0) and while it failed at the same point, it also seemebed to continue running the rest of the build steps, however the end result with the Chrome app was the same.

```
> coding-with-chrome@2.0.0 build-cwc-files E:\workspaces\git\coding-with-chrome-2.0.0
> node build/cwc_files.js

[14%] cwc.ui.Builder Compiling 51 soy files to C:\Users\Carlos\AppData\Local\Temp\closure-builder-JGkwf5M (0.0 sec)soynode: Compile error
 Error: spawn java ENOENT
[Soy Compiler Error] Failed for C:\Users\Carlos\AppData\Local\Temp\closure-builder-JGkwf5M:Error: Error compiling templates
[100%] cwc.framework.Runner [Done] (156.2 sec)
[100%] cwc.framework.Arduino [Done] (158.7 sec)
[100%] cwc.framework.TTS [Done] (164.1 sec)
[100%] cwc.framework.Sphero [Done] (166.9 sec)
[100%] cwc.framework.Ev3 [Done] (167.1 sec)
[100%] cwc.framework.simple.Loader [Done] (166.7 sec)

```

carlosperate, 2016-01-24:
Since this is related to the soy templates, this might be caused by not having Java installed on this computer. If this is the case, then the README should change the java requirement from optional to mandatory.

MarkusBordihn, 2016-01-25:
Normally the open Source version of java should be automatically installed.
It seems it worked fine for the normal compiler see "cwc.framework..." messages, but it failed for the soy template compiler.

At workaround you could install java to solve the issue, but I will try to adjust the "closure-builder" to automatically handling such case in the future.

carlosperate, 2016-01-25:
Do you mean like a bundled self-contained version of Java (like the Arduino IDE does)? I'd really prefer not to install the jre on my mail dev pc, but I might try it on a vm.

MarkusBordihn, 2016-01-25:
Yes, the closure compiler is using an bundled version if no java is available.
But the soy compiler is unfortunately not doing this.

johanlindberg, 2016-01-26:
A build error with the Soy template also happens on Mac OS X (10.9.5) Is it the same as you have reported @carlosperate ?

```
> node build/cwc_files.js

[14%] cwc.ui.Builder Compiling 52 soy files to /var/folders/r8/5k_vqzd911qflw_nqb9jrm700000gn/T/closure-builder-umWv4Fk (0.0 sec)soynode: Compile error
 Exception in thread "main" java.lang.UnsupportedClassVersionError: com/google/template/soy/SoyToJsSrcCompiler : Unsupported major.minor version 51.0
	at java.lang.ClassLoader.defineClass1(Native Method)
	at java.lang.ClassLoader.defineClassCond(ClassLoader.java:637)
	at java.lang.ClassLoader.defineClass(ClassLoader.java:621)
	at java.security.SecureClassLoader.defineClass(SecureClassLoader.java:141)
	at java.net.URLClassLoader.defineClass(URLClassLoader.java:283)
	at java.net.URLClassLoader.access$000(URLClassLoader.java:58)
	at java.net.URLClassLoader$1.run(URLClassLoader.java:197)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.net.URLClassLoader.findClass(URLClassLoader.java:190)
	at java.lang.ClassLoader.loadClass(ClassLoader.java:306)
	at sun.misc.Launcher$AppClassLoader.loadClass(Launcher.java:301)
	at java.lang.ClassLoader.loadClass(ClassLoader.java:247)

[Soy Compiler Error] Failed for /var/folders/r8/5k_vqzd911qflw_nqb9jrm700000gn/T/closure-builder-umWv4Fk:Error: Error compiling templates
```

carlosperate, 2016-01-26:
Not quite the same, mine fails as soon as it tries to launch java to compile the template, yours seems to be an issue with the Java version itself: http://stackoverflow.com/questions/10382929/how-to-fix-unsupported-major-minor-version-51-0-error

johanlindberg, 2016-01-26:
Right you are. There was a 1.6 JRE lurking behind `/usr/bin/java` on that machine. I've fixed that now hoping to get a clean build eventually.

carlosperate, 2016-06-07:
So are the soy templates now using the bundled version of Java?

ergelo, 2016-06-07:
@MarkusBordihn will know? I was going to add this to an FAQ, or do we want to keep this issue open until we are bundling Java in the soy compiler?

MarkusBordihn, 2016-06-07:
Unfortunately the bundled Java version is not longer maintained (since 3 years), so we are not able to use easily an bundled Java versions instead.

But I have create an feature request to use some online services instead.
https://github.com/google/closure-builder/issues/4

@ergelo 
As soon you have add this to an FAQ you could close this.
I have enabled the WIKI tab for this reason where we could add additional documentation.

closed 2016-10-13 by MarkusBordihn