google/coding-with-chrome #20: npm install fails on crouton

johanlindberg, 2016-01-24:
I'm running crouton on an Acer C720P and `npm install` fails. It seems to be because `closure-builder` depends on `fsevents` that apparently doesn't support Linux. I'm a bit confused about the "Skipping failed __optional__ dependency" message though. Any ideas on how to proceed?
```
(trusty)johan@localhost:/media/removable/SD Card/Projects/coding-with-chrome$ npm install closure-builder
coding-with-chrome@2.1.21 /var/host/media/removable/SD Card/Projects/coding-with-chrome
├─┬ eslint@1.10.3
│ └── glob@5.0.15 
├─┬ htmlhint@0.9.12
│ └── glob@5.0.15 
└─┬ npm-check-updates@2.5.6
  ├─┬ npm@3.5.3
  │ └── glob@5.0.15 
  └─┬ npmi@1.0.1
    └─┬ npm@2.14.16
      └── glob@5.0.15 

npm WARN optional Skipping failed optional dependency /chokidar/fsevents:
npm WARN notsup Not compatible with your operating system or architecture: fsevents@1.0.6
npm ERR! Linux 3.8.11
npm ERR! argv "node" "/usr/local/bin/npm" "install" "closure-builder"
npm ERR! node v0.12.9
npm ERR! npm  v3.5.3
npm ERR! path ../acorn/bin/acorn
npm ERR! code ENOSYS
npm ERR! errno -38

npm ERR! ENOSYS, symlink '../acorn/bin/acorn'
npm ERR! 
npm ERR! If you need help, you may report this error at:
npm ERR!     <https://github.com/npm/npm/issues>

npm ERR! Please include the following file with any support request:
npm ERR!     /var/host/media/removable/SD Card/Projects/coding-with-chrome/npm-debug.log
```

carlosperate, 2016-01-24:
The optional dependencies you are referring is coming from https://github.com/paulmillr/chokidar/blob/ffa1f8f70c10fa730f09e8149d6ee14040248c66/package.json#L41-L43 , but that one is being skipped, so shouldn't be a problem during installation.

This seems to be a related issue from npm: https://github.com/npm/npm/issues/10768

johanlindberg, 2016-01-24:
Thanks. Digging deeper I've concluded that `fsevents` had nothing to with it. It seems like the installer is trying to run a `symlink` command and my Linux installation won't have none of that.

johanlindberg, 2016-01-25:
Running `npm install --no-bin-links` seems to work. It makes sense, since the problem seems to be related to creating a symlink to a binary that doesn't exist in my installation.

closed 2016-01-25 by johanlindberg