google/coding-with-chrome #22: npm run build fails on crouton

johanlindberg, 2016-01-25:
I'm running crouton on an Acer C720P and `npm run build` fails. At first it failed because I didn't have a Java installation on my machine but after `sudo apt-get install default-jdk` that worked itself out nicely. However, now it fails with the below. Any ideas on how to proceed?

```
[ ... lots of removed output here ... ]
[100%] HTMLHint files [Done] (0.0 sec)
[14%] CodeMirror css Saving output to genfiles/external/codemirror/codemirror.css (0.9 sec)/var/host/media/removable/SD Card/Projects/coding-with-chrome/node_modules/closure-builder/build_compilers.js:375
      var message_info = message.match(message_reg);
                                 ^
TypeError: undefined is not a function
    at Function.<anonymous> (/var/host/media/removable/SD Card/Projects/coding-with-chrome/node_modules/closure-builder/build_compilers.js:375:34)
    at /var/host/media/removable/SD Card/Projects/coding-with-chrome/node_modules/closurecompiler/ClosureCompiler.js:289:21
    at ChildProcess.<anonymous> (/var/host/media/removable/SD Card/Projects/coding-with-chrome/node_modules/closurecompiler/ClosureCompiler.js:276:15)
    at ChildProcess.emit (events.js:110:17)
    at Process.ChildProcess._handle.onexit (child_process.js:1075:12)

npm ERR! Linux 3.8.11
npm ERR! argv "node" "/usr/local/bin/npm" "run" "build-extra-files"
npm ERR! node v0.12.9
npm ERR! npm  v3.5.3
npm ERR! code ELIFECYCLE
npm ERR! coding-with-chrome@2.1.21 build-extra-files: `node build/extra_files.js`
npm ERR! Exit status 1
[ ... and more removed output here ... ]
```

MarkusBordihn, 2016-01-25:
Its seems that crouton is limited in some ways, so that it will not passed backed the compiler messages as "string".
I will fix the "closure-builder" to explicit checking for the type to avoid this error.

But in general you could build "coding-with-chrome" on any platform, so you could build it on Mac, Windows or Linux and use it in Chrome OS. The compiled version in the "genfiles" folder is cross platform.

If I find the time I will check this more in detail on "crouton".

johanlindberg, 2016-01-26:
Ok. Making progress.

I just `npm update closure-builder --no-bin-links` since I noticed you made a few changes to the troublesome area in closure-builder yesterday. Now, closure-builder seem to work and `npm run build` fails silently. At first I thought the build had worked fine but loading the extension in Chrome didn't work. It was stuck at 1% Building the Coding with Chrome UI. Inspecting the Editor view revealed the following Javascript errors:
```
chrome-extension://ffiiemmmlmmofcoihjlobpkjpgmcclak/external/codemirror/addons.js Failed to load resource: net::ERR_FILE_NOT_FOUND
chrome-extension://ffiiemmmlmmofcoihjlobpkjpgmcclak/external/codemirror/modes.js Failed to load resource: net::ERR_FILE_NOT_FOUND
chrome-extension://ffiiemmmlmmofcoihjlobpkjpgmcclak/js/cwc_ui.js Failed to load resource: net::ERR_FILE_NOT_FOUND
editor.js:30 Uncaught ReferenceError: cwc is not defined
```

alexbaumgertner, 2016-02-16:
Maybe I have the same issue. Mac Book mid 2010, El Capitan
node: v5.5.0
npm: 3.5.1
```
> coding-with-chrome@2.2.5 build-cwc-files /Users/alexbaumgertner/Projects/coding-with-chrome
> node build/cwc_files.js

[14%] cwc.ui.Builder Compiling 52 soy files to /var/folders/wh/n_q0s1t956lf0zypsckp92pm0000gn/T/closure-builder-dEQHWMx (0.0 sec)soynode: Compile error
 Exception in thread "main" java.lang.UnsupportedClassVersionError: com/google/template/soy/SoyToJsSrcCompiler : Unsupported major.minor version 51.0
	at java.lang.ClassLoader.defineClass1(Native Method)
	at java.lang.ClassLoader.defineClassCond(ClassLoader.java:637)
	at java.lang.ClassLoader.defineClass(ClassLoader.java:621)
	at java.security.SecureClassLoader.defineClass(SecureClassLoader.java:141)
	at java.net.URLClassLoader.defineClass(URLClassLoader.java:283)
	at java.net.URLClassLoader.access$000(URLClassLoader.java:58)
	at java.net.URLClassLoader$1.run(URLClassLoader.java:197)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.net.URLClassLoader.findClass(URLClassLoader.java:190)
	at java.lang.ClassLoader.loadClass(ClassLoader.java:306)
	at sun.misc.Launcher$AppClassLoader.loadClass(Launcher.java:301)
	at java.lang.ClassLoader.loadClass(ClassLoader.java:247)

[Soy Compiler Error] Failed for /var/folders/wh/n_q0s1t956lf0zypsckp92pm0000gn/T/closure-builder-dEQHWMx:Error: Error compiling templates
```

After I installed java [by brew](http://stackoverflow.com/questions/24342886/how-to-install-java-8-on-mac)
```
$ brew update
$ brew tap caskroom/cask
$ brew install Caskroom/cask/java
```
it seems working well:
[image]
[image]

closed 2016-03-28 by MarkusBordihn