google/coding-with-chrome #24: Possible pypyjs integration ?

MarkusBordihn, 2016-02-03:
Maybe we should integrate pypyjs to support Python as input language as well.

See https://github.com/pypyjs/pypyjs

carlosperate, 2016-02-03:
Coincidentally I was looking into this today during my lunch break for something different, one very simple requirement I needed was to be able to use `time.sleep()` (as that is one of the first steps in blinking an LED), and after a very brief look around into these three implementations I only found skulpt to work as expected.

This might be related at all to the requirements for this projects, but I though I'd mention it as a simple example of how something we'd think as trivial can be such a deal breaker sometimes.

carlosperate, 2016-02-03:
Also! it would be very interesting to identify which one would allow you to step through the code, to be able to do the same type of thing the "JS Interpreter" blockly example does.

MarkusBordihn, 2016-02-04:
In general it would not be possible to do the step through the code thing very easily in an Chrome app, because the code is executed in an sandbox (webview).

So the executed code is not able to tell you which is currently executed, nor the UI knows exactly what the sandbox is executing.

MartenvanWezel, 2016-02-04:
As far as I can tell all of the above implementations have their upsides and downsides. As I understand it, brython implements Python 3.x while skulpt is most similar to Python 2.2 (while 2.7 is the gold standard at the moment). pypyjs seems to be the most compliant and guess what- the slowest.

Then again, it seems that due to various reasons there is no sleep() equivalent in javascript due to its single threaded nature (source: http://reeborg.ca/docs/en/reference/weakness.html) so indeed implementing it is not as easy as it sounds - although I agree for educational purposes it is important. This thread explains it in more detail: https://groups.google.com/forum/#!topic/brython/UAZADu6Yv_g  - but in short it seems to me that the only way to 'emulate' sleep in javascript is to create a loop and breaking out of it after some time. Problem: that shoots the CPU to 100% 

Bottom line, I currently see no obvious 'winner' between the implementations themselves, so instead I would suggest looking at them in the meta-level, and picking one based on project activity/maturity, ease of implementing etc. If all that fails... toss a coin? 

But yes, I would consider it very valuable to support Python.

MarkusBordihn, 2016-02-04:
I just checked Brython and is supports time.sleep as well after "import time".
But as already mention it puts the CPU to 100% so in the worst case the sandbox will kill it.

MartenvanWezel, 2016-02-04:
I assume that this killing won't happen after 1 second, so sounds to me it will be fine day-to-day,

aroberge, 2016-02-29:
_Since someone mentioned my site and a reply I wrote..._ 

If all that is needed is a "step" method, this can be implemented fairly easily without a sleep() method: all that needs to be done is to record the result of the program (after doing some pre-processing) and play it back.  This is what I have done with Reeborg whose documentation is linked in a previous comment.  You can see the result with a Blockly integration by going to

http://reeborg.ca/reeborg.html?lang=en&mode=blockly-py

clicking on the "Keep editor visible" checkbox and dragging the editor out of the way.  For a demo, I suggest creating a simple program using combinations of "move()" and "turn_left()" functions.  When clicking on the "run" button (near the top left), the program will be executed one instruction at a time with the code in the editor being highlighted; the speed of the animation can easily be adjusted (using the "think()" function) and can even be played back forward and backward (the latter available via the additional options menu).    Instead of the line of code, one could presumably highlight the block being executed instead.  (I may look into doing this "soon".)

A more robust method could likely be implemented using a Python-debugger (https://github.com/amrdraz/python-debugger) designed explicitly to work with Brython; I am simply aware of its existence and have not attempted to use it.

Note that the link to "Reeborg's World" given above is for a newer development version that is incomplete and likely buggy.

MarkusBordihn, 2016-07-19:
Thanks for all of your feedback on this, very appreciated.

I just implemented an basic test implemented at the python branch for testing:
https://github.com/google/coding-with-chrome/tree/python

Seems to work fine with generic Python examples.

closed 2016-07-19 by MarkusBordihn