google/coding-with-chrome #29: Sphero relative blocks

MartenvanWezel, 2016-02-28:
Currently, we can tell sphero to go direction <x> degrees, where (say) 90deg is 'right'. 
It would be very good if we can also implement a block that allows us to program the sphero in a more relative way, i.e. "turn x degrees" and then "move forward". 

Reason is that it is a lot more easy to use variables and loops ("turn x degrees, move forward, repeat a few times")

MarkusBordihn, 2016-02-28:
Sphero is designed to be absolute, so even it would help in some cases its not really feasible because its not supported by the hardware itself.

So it would be only virtual and would cause sooner or later an mismatch between the real position and the virtual position.

closed 2016-02-28 by MarkusBordihn

MartenvanWezel, 2016-02-28:
Is there at least some way to programmatically modify the direction? Meaning doing something like a few blocks like move to direction <result>, and then doing some math with the direction?

it would be good to be able to make a loop where the direction depends on a variable.

MarkusBordihn, 2016-02-28:
Could you please provide some example code or an example task for this ?

At the moment I see no real use case where the absolute position can't be used.

MartenvanWezel, 2016-02-28:
How would you draw (drive) a cube (or pentagram) a repeat block? Of course
you could spell out all the angles, but it would get silly very quick.

EV3: repeat 4 times, drive 200, turn 90, drive 200
Sphero: ? I tried count with i from 0 to 360 by 90 but couldn't put 'i'
into the "roll for x seconds with speed and heading" block.

MarkusBordihn, 2016-02-28:
Please check the examples of http://www.sphero.com/education which covering such use cases.

I will maybe translate some of them, but again I don't see any need for such a function.