google/coding-with-chrome #59: Switch to Web Bluetooth

beaufortfrancois, 2016-05-12:
Once https://bugs.chromium.org/p/chromium/issues/detail?id=577170 is addressed, you might want to give a try to the [Web Bluetooth API](https://developers.google.com/web/updates/2015/07/interact-with-ble-devices-on-the-web).

MarkusBordihn, 2016-05-12:
We already checked this option, but unfortunately its only supporting BT 4.0 and higher devices.
The EV3 and Sphero 2.0 are only support BT 2.x so currently it would not be an options for these devices.

beaufortfrancois, 2016-05-12:
Makes sense!

beaufortfrancois, 2016-08-04:
Why is the low_energy permission added in the manifest then?
https://github.com/google/coding-with-chrome/blob/713bbe27d1fe4a481a80640784388eff41784781/app/manifest.json#L52

MarkusBordihn, 2016-08-04:
Mainly for testing reasons.

Currently we are not supporting Bluetooth low energy device, because the needed protocol implementation is missing under https://github.com/google/coding-with-chrome/tree/master/src/protocol .

But supporting Bluetooth LE is something which is on our road-map.
Furthermore we also working on to make Coding with Chrome cross compatible, which mean it could be used in an normal browser besides the Chrome App version.

beaufortfrancois, 2016-08-04:
That would be awesome! 
Feel free to ping me in the future if you're having any issues with the Web Bluetooth API.

closed 2016-08-04 by beaufortfrancois