google/coding-with-chrome #60: Upon refocussing to CWC window, moving window doesn't always work

MartenvanWezel, 2016-05-26:
Having some trouble reproducing this dependably with a recipe but very very often

1/ Switch to another window (say a browser window)
2/ Click on CWC window
3/ Try to drag the CWC window around (click and hold blue area on top of the window, move mouse)

-> Often it doesn't work, window doesn't move, and I have to click and drag a couple of times before I can move the window

This feels a bit more like a Chrome bug than a CWC bug but it's worth mentioning.

MarkusBordihn, 2016-05-26:
No all parts of the blue are "drag-able", there is an reserve area for the text on the left which is not drag-able. I will try to make this area more visible to avoid such confusions.

MartenvanWezel, 2016-05-26:
Ah.. that's.. confusing :) - but yes you seem to be correct. 

I suppose it makes sense in the main mode where clicking on that area modifies the filename (although I would argue that renaming the file that way is not very common in most mac/windows apps)

Frankly I would remove the rename function (through clicking the window title) everywhere.

MarkusBordihn, 2016-05-27:
Its the standard for most of the Google Apps like docs, spreadsheet, ...

See also:
https://getmdl.io/components/index.html#textfields-section

closed 2016-05-27 by MarkusBordihn