google/coding-with-chrome #64: Unable to rename variable or create new

mrpoponz, 2016-06-01:
Using blocky on version 3.5.16 in chrome on windows and an asus c201 chromebook.

**To replicate**
Beginner - Blocky -> Drag a variable -> Rename or new

**Behavior**
Nothing happens

**Expected behavior**
Input box on screen

**Notes**
I've also tried https://blockly-demo.appspot.com/static/demos/interpreter/index.html and it works fine so it may not be a browser problem.

MarkusBordihn, 2016-06-01:
Blockly is using "window.prompt" for these actions which is blocked for Chrome Apps and Extensions by default.

bdolicki, 2016-06-02:
Please prioritize this. It essentially prevents users from using variables (or rather, more than one variable called "item"). That makes a lot of programs impossible to write :)

MarkusBordihn, 2016-06-02:
Reference:
https://github.com/google/blockly/issues/393

MarkusBordihn, 2016-06-02:
@MartenvanWezel 
Please test and update the bundled version on the Chromestore to fix this issue.

closed 2016-06-02 by MarkusBordihn

MartenvanWezel, 2016-06-02:
Nope. Build from scratch failed. (master branch)

[14%] cwc.ui.Builder Compiling 56 soy files to /var/folders/k8/b9b7qkln48j74xb5z[15%] cwc.ui.Builder Compiled 56 soy files to /var/folders/k8/b9b…sure-builder-N[16%] cwc.ui.Builder Compiling Closure Files (1.8 sec)[Closure Compiler Error] 1 errors for genfiles/js/cwc_ui.js:node_modules/blockly/blocks/lists.js:68: WARNING - missing require: 'Blockly.Mutator'
    this.setMutator(new Blockly.Mutator(['lists_create_with_item']));

MartenvanWezel, 2016-06-02:
node_modules/blockly/core/field_variable.js:29: ERROR - required "Blockly.Events.Change" namespace never provided
goog.require('Blockly.Events.Change');
^

1 error(s), 68 warning(s)

MarkusBordihn, 2016-06-03:
Travis CL said everything is fine now:
https://travis-ci.org/google/coding-with-chrome

Please make sure to run "npm run update" before, to fetch the latest changes.

MarkusBordihn, 2016-06-03:
Fixed version if live now.

mrpoponz, 2016-06-03:
Would this be the same for "window.prompt"?  Under Text -> Prompt for "Text" with message.

MarkusBordihn, 2016-06-03:
@mrpoponz 
Thanks for the hint. This would be the same problem, but it could be solved on our end instead of the Blockly side.

Will handle all of these in #68.