google/coding-with-chrome #69: [user feedback] "unknown error" when trying to connect to EV3

MartenvanWezel, 2016-06-08:
"ev3 bluetooth
chromebook bluetooths to ev3, but coding with chrome gives 'unknown error' when trying to bluetooth and does not work"

MarkusBordihn, 2016-06-08:
Please provide some more details like "console messages", "chrome log", ... about this issue. "Unknown error" is very generic and could mean everything or nothing.

Shonkk, 2016-06-15:
Add Bluetooth Device Console Message: "An unknown error occurred  trying to connect to "00:16:53:4A:85:BA"
The Bluetooth icon in the top right is grey in colour and when it is clicked, nothing happens.
Do you have a link that describes how to get a chrome log, I have never done that before.
Thanks in advance

MarkusBordihn, 2016-06-15:
@Shonkk
Could you please provide the following additional information:

- Type / Model / Name of Chromebook ?
- Type of EV3 (standard or education) ?
- Any specific modifications to the EV3 like custom image, linux image, ev3dev ?
- EV3 firmware version ?
- EV3 Bluetooth settings (e.g. no iPhone Mode, Visible) ?

Thanks and Greetings,
- Markus B.

Shonkk, 2016-06-15:
Thank you for your patience with me!
Chromebook  model - Samsung ARMv7 Processor rev 4 (v71)
EV3 - education
No modifications to the EV3 at all
EV3 brick HW: V0.60,   EV3 brick FW: V1.07H
EV3 bluetooth settings, Visibility, bluetooth, iphone/ipad/ipod all with checkmarks.

In playing around with it, I am noticing that when I bluetooth to the EV3 using the chromebook menu bluetooth, it appears to connect for about 5 seconds and then loses the connection (although no message is given - the device name goes BOLD for 5 seconds and then goes back to normal font).  So maybe the bluetooth is not connecting probably from the chromebook not the cwc app?

MarkusBordihn, 2016-06-15:
Thanks for the details, will try to reproduce the issue.
Does it makes any difference if you disable the "iphone/ipad/ipod" mode ?

Shonkk, 2016-06-15:
nope, I tried that

MarkusBordihn, 2016-06-15:
Thanks. Coding with Chrome is using the Bluetooth driver of the Chrome book.
The described issue sounds like an incompatible driver / hardware combination.

@MartenvanWezel 
Please try to get the same Chromebook model any try to reproduce the issue.

Shonkk, 2016-06-15:
Same issue with our other chromebooks:
model : Dell 69
Chrome version: 51.0.2704.79

I am a teacher with a school board.  Many of our schools only have chromebooks and we just received 100 EV3s.  We are trying to find a way to use them with chromebooks so are very excited to find this app.  Thanks for your efforts.

MarkusBordihn, 2016-06-15:
Unfortunately it seems like an issue on the Chrome OS level see:
https://bugs.chromium.org/p/chromium/issues/detail?id=227214

Because there is not a lot we could do on the application level,
I recommend to open a bug with the Chrome OS team.

@MartenvanWezel 
Would be great if you could try to reproduce the issue to see if we could do something to help here.

MarkusBordihn, 2016-10-13:
Unfortunately we were unable to reproduce the issue with our hardware.
But we are working on improving the Bluetooth connection screen, which should allow an easy troubleshooting in the feature.

closed 2016-10-13 by MarkusBordihn