google/coding-with-chrome #78: EV3 Joystick Input

stevenpaulr, 2016-08-19:
I'm using the old RobotC NXT curriculum, but I want to use Coding with Chrome and EV3s instead of RobotC. The last challenge is to program the robots to respond to joystick commands. I would like to be able to do the same thing with the EV3s.

MarkusBordihn, 2016-08-19:
Could you please share an example which explains your need more in detail?

At the moment the IR remote control is currently not implemented, but you are already able to control the EV3 over the keyboard or the ui.

stevenpaulr, 2016-08-23:
I want to be able to code the input like this:

  getJoystickSettings(joystick);     // update buttons and joysticks
   if(joystick.joy1_Buttons == 32)    // if Button 6 is pressed on joy1:
  {
    motor[motorA] = 50;                // motorA is run at a power level of 50
  }

or this:

getJoystickSettings(joystick);     // update buttons and joysticks
motor[motorB] = joystick.joy1_x2;  // motorB's powerlevel is set to the right stick's current x-value

MarkusBordihn, 2016-08-23:
Ok thanks for your feedback. This should be feasible over the Gamepad API (https://w3c.github.io/gamepad/).
Will check this more in detail and will create an example for this.

closed 2018-03-19 by MarkusBordihn