google/coding-with-chrome #82: Switch From Code to Blockly

davidribyrne, 2016-09-18:
Users can currently switch from Blockly to code. It would be nice to go back the other way without losing changes. Code.org (https://github.com/code-dot-org/code-dot-org) has this feature. Perhaps it can be ported over. #

MartenvanWezel, 2016-09-18:
Are you sure that toggling forward and back works perfectly? As far as I understand it, it's not possible to just translate arbitrary javascript into blockly, there aren't blockly equivalents for all types of code..

davidribyrne, 2016-09-18:
There may be some limitations, but I haven't seen any with the type of code that most students write. https://code.org/educate/applab

MarkusBordihn, 2016-09-19:
What they are doing is not to "understand" the code, instead they using an "pseudo" block for every definition which is unknown.
This will work in most cases, but would mean that variables and functions are not detected and not part of the available variable or function list (features are not used by code-dot-org).

We will check if this could be easily done or if it will break something.

MarkusBordihn, 2018-09-14:
Checked this use case with the blockly team.
Unfortunately with es6 and async, promise and all other chainable objects this is not really feasible.
So it would work for basic stuff but not for the more advanced stuff.

closed 2018-09-14 by MarkusBordihn