google/coding-with-chrome #88: Translation to Swedish.

therealeldaria, 2016-10-06:
Hello I would like to make a Swedish translation to enable pre-school and lower school children to use Coding with Chrome.
I found that Blokly is already translated using the https://translatewiki.net/ but not Coding with Chrome.

I found that some parts have partial German translations and also I think Korean, and these are the options that are possible in the app. 
But if you could point me to what files I need to modify, then I will submit Swedish translation.

MarkusBordihn, 2016-10-13:
Hi,
thanks a lot for your interest. The translation files are available at the following location:
https://github.com/google/coding-with-chrome/tree/master/src/locales

Basically you need to create a copy of the "en" folder as "se" folder and adjust the text in the "translation.js" file.

Some additional documentation:
https://github.com/google/coding-with-chrome/blob/master/doc/I18N.md

therealeldaria, 2016-10-13:
Ok cool, I don't have any experience with git, but I will try to read up on how to do it and submit the Swedish translation. It does not seem to be so much in that file so should be able to get time for it next week.

MarkusBordihn, 2016-10-14:
Basically you don't need to learn git for an single file, instead you could use the GitHub Editor see: https://help.github.com/articles/editing-files-in-another-user-s-repository/

Let me know, if I should prepare the file so that you could simple edit the file over GitHub.

therealeldaria, 2016-10-14:
I figured out how to fork it to my own account and then download it, so I have it locally now. I could not figure out how to copy the folder or even just create a new folder in the locales by using the browser interface.
But for now since I have it locally I already started translating it. I have some other things this weekend so will see how much I will get done in between, but so far I got to line 126.

I did notice that there are some errors in the English version as well. One was on line 38: 
"Automatic reloads the preview after an change on the editor content."

For example: "an" should be used in front of a words that is or sounds like a vowel, and "a" should be used in front of words that is or sounds like a consonant.

So there are a few ways one could write it, my suggestion would be to change it to.
"Automatically reloads the preview after a change of the content in the editor"
Another way it could be written would be:
"Automatic reload of the preview after a change in the editor content."

MarkusBordihn, 2017-03-29:
We reviewed all of the text and replaced the language system.
Let me know if there is still interest to provide an Swedish translation.

New language file:
https://github.com/google/coding-with-chrome/blob/master/locales/en/translation.js

therealeldaria, 2017-03-29:
> master/locales/en/translation.js

Hello Markus.

Yes I would still like to help out with this, I started on the previous
one, but think I did something wrong when I started, since I have it in my
own profile so was not sure If I could get it back into the main branch.
It is here:
https://github.com/therealeldaria/coding-with-chrome/blob/master/src/locales/se/translation.js

It looks like there are a lot of changes, almost double the lines, guess it
is better if I start over or can what I done so far be merged in?
And will I be able to get the translation I make into the main branch in
the way I did it?

/Brian

MarkusBordihn, 2017-03-29:
Hi Brian,
basically its not issue if you have this in your own branch the pull request could be done from any branch.

Unfortunately, because of all new changes, I recommend to backup your old file, deleting your branch and forking the current branch.

After this create the file:
https://github.com/google/coding-with-chrome/blob/master/locales/se/translation.js

and use the English version as an kind of template.

You don't need to translate everything at once, its fine if you start with some basic words and adding other later.
Keep in mind that only translated words should be included in the file

SetCodesToFire, 2018-01-21:
Can I do this?

therealeldaria, 2018-10-28:
Hello, Long time. So In finally got around to start the Swedish translation.
So I saw that the language system changed again, but I have made a new fork, made a copy of the eng folder in locales and named it swe.
I have translated all files except the translation.js as it says it is deprecated.
I also added Swedish to i18n_mapping.js.
And when I build the application I can pick Swedish, but it does not switch language when I pick Swedish so I think there is something missing.

MarkusBordihn, 2018-10-31:
@therealeldaria 
Please open a pull request so that I could review the changes and fix anything which is missing.
Thanks. ;)

MarkusBordihn, 2018-11-02:
Hi Brian,
I fixed two missing build rules in the PR and it should be working now (master branch).
Thanks a lot for your contribution.

closed 2018-11-02 by MarkusBordihn

therealeldaria, 2018-11-02:
Noticed I did not change author on all Swedish files, not important thought. ;-)

MarkusBordihn, 2018-11-02:
Feel free to adjust the author for the relevant files and I will merge the PR right away.

MarkusBordihn, 2018-11-02:
Unfortunately there is currently no automatic way to do something like this.
But this sounds like a good idea and I will let some of our team members explore possible options more in detail.

@MartenvanWezel 
Could you please check if we could implement an notification system for translators to get informed if the main languages files are adjusted, maybe https://translatewiki.net/ is and option?

MartenvanWezel, 2018-12-01:
[+keisuke - what is your take on this?