google/coding-with-chrome #93: Line numbers of errors?

MartenvanWezel, 2016-11-24:
for example:
1/ advanced > programming > javascript > circle animation
2/ create a typo somewhere
3/ would be nice to see a line number of the error?

Somehow I thought we already had this feature, but it leaves me in the dark WHERE the error is.

[image]

(the fact that the line with the error highlighted is just my highlighting, the editor itself doesn't show it)

MarkusBordihn, 2016-11-25:
In general there is an different between syntax error and runtime errors.
For the runtime errors we don't have any line numbers, because the executed code is not 1:1 the same as the visible code.

If you copy the same line outside of the function it will become an syntax error which will be detected by this function.

MartenvanWezel, 2016-11-25:
I see - that does make sense from a 'code structure' level, but it would be nice in some way to do this inside functions that can only be interpreted at runtime too. You mentioned we might be able to implement this at heavy processing cost (so it should be enabled optionally?) but it would be a cool one so let's leave this as a feature request?

closed 2016-11-25 by MartenvanWezel