google/coding-with-chrome #96: "Sleep" block

MartenvanWezel, 2017-01-12:
I'm aware this is harder than it seems, but it would be very good to have a block that has the blockly code wait for a certain amount of ms/sec

MartenvanWezel, 2017-03-20:
This has been requested over and over again, in particular in blockly context. (so presumably the people requesting it don't "understand" that sleep doesn't exist in JavaScript. 

Would it make sense to build something that does like:

function sleep(milliseconds) {
  var start = new Date().getTime();
  for (var i = 0; i < 1e7; i++) {
    if ((new Date().getTime() - start) > milliseconds){
      break;
    }
  }
}

Clearly this will make the CPU go nuts looping while waiting, but frankly I think it might be better than not having a sleep function...

MarkusBordihn, 2017-03-20:
Could you please describe the detail use-case here ?

In most of the cases its better to use setInterval or setTimeout, but there is not really the need for any "sleep" block.

We should not provide an false "sleep" block instead we should explain the alternatives.
Furthermore with ES6 and NodeJs most of the JavaScript runs parallel so the above function would not work in such cases, besides it could crash the machine. ;)

MartenvanWezel, 2017-03-20:
The use case in it's simplest form is e.g.

1/ Write "3"
2/ Wait a second
3/ Write "2"
4/ Wait a second
5/ Write "1"
6/ Wait a second
7/ Move Sphero Forward max speed

Just... wait. Not do everything as soon as possible but give humans time to notice what's happening =)

MarkusBordihn, 2017-03-20:
This use-case describe more an "wait" block instead of the "sleep" block.
We have an StackQueue function with an "wait" functionality, which could be maybe implemented as block, or we are providing and "delay" block instead.

Interesting discussion about this:
https://groups.google.com/forum/#!topic/blockly/5xbDQ8FQzbI

MartenvanWezel, 2017-03-20:
Sure, I'm using the terms "sleep" and "wait" interchangeably. It's just that there are many things that go too fast in CwC (especially in 'educational' context) that need to be slowed down by inserting some step. 

I'm happy to accept this is sort-of an antipattern in raw JavaScript but at least in blockly I think it is important to have this.

MarkusBordihn, 2017-05-03:
https://developer.mozilla.org/en-US/docs/Web/JavaScript/Reference/Operators/await

Example:
...
function addDelay(delay) {
  return new Promise(resolve => {
    setTimeout(() => resolve(), delay);
  });
}

async function f1() {
  await addDelay(1000);
  console.log('Hello');
  await addDelay(1000);
  console.log('World');
  await addDelay(1000);
  console.log('1');
  await addDelay(1000);
  console.log('2');
};

f1();
...

MartenvanWezel, 2017-05-08:
Sounds Good :) - Make a block out of it? ;)

MarkusBordihn, 2017-05-08:
As already mentioned it makes no sense for the robot implementation, because the robots are not supporting any "wait" command or state.

Which mean the only usable place would be the "Basic" blocks.

closed 2017-07-24 by MarkusBordihn