google/coding-with-chrome #99: What's new splash page.

MartenvanWezel, 2017-01-26:
I'd love an option for me (as PM) to be able to manually modify a file in the package that will be displayed directly.

- Default file, just maybe write "CWC has been updated to x.y.z
- On first startup after updating
- Allow me to modify the default file with a more customized message ('new features!')
- With a switch to disable any popups (if teachers don't want their students confused)

We had some feedback that people think we don't update whereas we do quite often IMO.

MarkusBordihn, 2017-01-26:
The more prefered options would be an changelog.
You could simple create an CHANGELOG.md file with the relevant milestones for the UI.

MartenvanWezel, 2017-02-13:
OK, but will the behavior I described happen? (every new version, display the changelog once on new startup?)

MarkusBordihn, 2017-02-14:
I would make this optional for the user in the settings menu.
So that the user could decided if they want to see this with every update or optional over the info menu.

MartenvanWezel, 2017-02-14:
Yup, like I said, we don't want to nag people if they're not interested in the changes.
One approach would be: (attached image)

[image]

MartenvanWezel, 2018-05-10:
@TioBorracho Are you able to take this one?

MartenvanWezel, 2018-05-10:
Some further detail:

Git has a decent commit log, and Markus&team have been pretty good at maintaining a decent level of detail: 

`git log --pretty=format:'%h,%at,"%s"' (https://mirrors.edge.kernel.org/pub/software/scm/git/docs/git-log.html#_pretty_formats)
`

But, in some cases, for some timestamps, I would want to create a "special announcement". For example:

Current log: (we probably need to use full commit hash here, but for brevity)

```
5e77755,Fri, 4 May 2018 22:31:54 +0200,"Added cover for unit tests."
ff6ae65,Fri, 4 May 2018 18:36:05 +0200,"Update .travis.yml"
cc482b6,Fri, 4 May 2018 18:11:12 +0200,"Added "--include-all-sources" for nyc to find all relevant files."
7832e84,Fri, 4 May 2018 18:02:32 +0200,"Added nyc test coverage reporter."
28f9f92,Fri, 4 May 2018 17:09:58 +0200,"Optimized build rules and test files for better performance. Added Test coverage for Code Climate."
eabc68a,Fri, 4 May 2018 07:12:57 -0600,"Added mode tests for tutorial and tour (#160)"
ebb6b7c,Fri, 4 May 2018 01:37:26 +0200,"Replaced javascript resizing with css resizing for better performance."
b5ebc0b,Wed, 2 May 2018 23:21:25 +0200,"Simplified sidebar content handling."
f760d1b,Wed, 2 May 2018 18:03:54 +0200,"Cleanup compiler warning messages."
74a6e9a,Wed, 2 May 2018 12:39:54 +0200,"Fixed and re-enable automated mode tests."
f78a83e,Wed, 2 May 2018 01:59:48 +0200,"Added editor tab support for supporting different views and multiple files in one project."
db5787b,Mon, 30 Apr 2018 20:13:58 +0200,"Merge branch 'adamcarheden-tutorial-content'"
8d3281d,Mon, 30 Apr 2018 20:13:32 +0200,"Better implementation of the Tutorial and Tour instance."
27e07a1,Mon, 30 Apr 2018 01:18:34 -0600,"Merge remote-tracking branch 'upstream/master' into tutorial-content"
134c9a7,Mon, 30 Apr 2018 02:15:21 +0200,"Improved i18n library and switch to promises instead of callbacks."
4f89e71,Mon, 30 Apr 2018 02:55:36 +0900,"fix typing (#158)"
6a97dc3,Sun, 29 Apr 2018 01:52:07 +0200,"Added nice loading screen with simple svg animation."
fbe1c6a,Fri, 27 Apr 2018 17:46:07 +0200,"Removed seperated loading screen for better web app support. Created new loading screen instance which will b
e also used for the web version."
65f2cf6,Thu, 26 Apr 2018 17:47:31 +0200,"Update README.md"
0a4ea60,Wed, 25 Apr 2018 17:56:03 +0200,"Fixed issue #156 and added additional icons for the sections."
e19e53d,Tue, 24 Apr 2018 22:40:41 -0600,"Merge remote-tracking branch 'upstream/master' into tutorial-content"
```

Let's imagine that:
- The user's version of CWC is at 5.4.251, built on April 25.
- We have a "RELEASE.md" file, that specifies, e.g.: (obviously in MD format)

`db5787b,"30 Apr","Major Release Update! We now have tutorial content!"`

Then, on May 4, Tony builds the new version. 

What I would like the code to do:

1/ Start up
2/ Check the release hash and date of current version
3/ Check storage (file? cache?) which version CwC was when it was run last
4/ If 2 and 3 differ, then this is the **first run of the new version** - time for action
5/ Check all release hashes between $oldversion and $currentversion
6/ Check the RELEASES.md file - does any of the hashes from #5 match a hash in #6 - e.g. is there a major announcement to report?
7a/ If major announcement: (check if announcement pop-ups are enabled in settings), but if yes, then DISPLAY ANNOUNCEMENT! ("Major Release Update! We now have tutorial content!")
7b/ If no major announcement, Just display the updated message (as per the mockup above)

This means:
- Implement the settings item, to enable(default) or disable update pop-ups
- Implement the 'updated' display for minor updates
- Implement the 'NEW STUFF OMGOMGOMG' pop-up for major updates

Edge case: perhaps 2+ major announcements have happened since the user last updated CWC. Display them all somehow.

TioBorracho, 2018-05-10:
Regarding these:

> - Implement the 'updated' display for minor updates
> - Implement the 'NEW STUFF OMGOMGOMG' pop-up for major updates

Should they be the same popup, but it will just not open automatically if there are not NEW STUFF updates?

MartenvanWezel, 2018-05-10:
btw all of the above is just a suggestion, happy to implement differently (e.g. feel free to suggest a better way for me to deliver a 'more detailed, official announcement')

MarkusBordihn, 2019-02-11:
The first basic version is implemented, so I assume this could be closed.

closed 2019-02-11 by MarkusBordihn