google/coding-with-chrome #111: Support for Python turtle library

milesberry, 2017-05-28:
Really impressed by what you've done here. 

Lots of secondary schools in England are defaulting to Python as their main teaching language for [the new computing curriculum](https://www.gov.uk/government/publications/national-curriculum-in-england-computing-programmes-of-study/national-curriculum-in-england-computing-programmes-of-study#key-stage-3) and accompanying qualifications, and I can see this being of great use where they are reluctant to install locally or are using Chromebooks for delivery.

Many teachers are finding elementary turtle graphics a great way to scaffold the transition from Scratch (popular in primary schools) to Python - pupils can write equivalent programs in each as they start to master Python syntax - see [this article in Hello World](http://s3-eu-west-1.amazonaws.com/rpi-magazines/issues/full_pdfs/000/000/002/original/HelloWorld01.pdf#page=46), and [this academic paper](https://pdfs.semanticscholar.org/00bd/b2fda5f662a0efdd217d22b2ebeeb72b3e53.pdf). Thanks to Papert's Logo work, turtle graphics has a long history in computing education: you've good support for turtle graphics in Pencil Code, but this is of limited use for schools wanting to scaffold pupils' learning in Python.

Coding with Chrome has much to recommend it, but for now I fear it's going to lose out (at least in introductory work in English schools) to tools like trinket.io and repl.it which provide Python turtle support.

MarkusBordihn, 2017-05-28:
Thanks a lot for the detailed feedback. The mentioned examples are based basically on the same Python framework, which mean that the basic turtle function will work in Coding with Chrome as well.

The following example code from trinket.io should work without any problem:
```python
#!/usr/bin/python2.7
from turtle import *

# Create a turtle named Tommy:
tommy = Turtle()
tommy.shape("turtle")
tommy.speed(7)

def draw_circle(turtle, color, size, x, y):
  turtle.penup()
  turtle.color(color)
  turtle.fillcolor(color)
  turtle.goto(x,y)
  turtle.pendown()
  turtle.begin_fill()
  turtle.circle(size)
  turtle.end_fill()
    
def draw_triangle(turtle, color, size, x, y):
  turtle.penup()
  turtle.color(color)
  turtle.fillcolor(color)
  turtle.goto(x,y)
  turtle.pendown()
  turtle.begin_fill()
  for i in range (3):
    turtle.forward(size*3)
    turtle.left(120)
  turtle.end_fill()
  turtle.setheading(0)
    
def draw_square(turtle, color, size, x, y):
  turtle.penup()
  turtle.color(color)
  turtle.fillcolor(color)
  turtle.goto(x,y)
  turtle.pendown()
  turtle.begin_fill()
  for i in range (4):
    turtle.forward(size*2)
    turtle.left(90)
  turtle.end_fill()
  turtle.setheading(0)
  
def draw_star(turtle, color, size, x, y):
  turtle.penup()
  turtle.color(color)
  turtle.fillcolor(color)
  turtle.goto(x,y)
  turtle.pendown()
  turtle.begin_fill()
  turtle.right(144)
  for i in range(5):
    turtle.forward(size*2)
    turtle.right(144)
    turtle.forward(size*2)
  turtle.end_fill()
  turtle.setheading(0)

# Draw three circles:
draw_circle(tommy, "green", 50, 25, 0)
draw_circle(tommy, "blue", 50, 0, 0)
draw_circle(tommy, "yellow", 50, -25, 0)

# Write a little message:
tommy.penup()
tommy.goto(0,-50)
tommy.color("black")
tommy.write("Teach With Code!", None, "center", "16pt bold")
tommy.goto(0,-80)
```

The only feature which is currently missing is the "multiple" file support.
Which mean that all of the python code needs to be in one file instead of several files.
Let us know if this will work in this case or if you are missing any other specific features.

milesberry, 2017-05-28:
Thanks! That works. 

I *did* try to get turtle graphics working before posting, but clearly had done something wrong. So sorry to have bothered you.

closed 2017-05-28 by milesberry

MarkusBordihn, 2017-05-28:
Perfect. No worries, it was great feedback and it shows that we need to improve some parts and that we should supporting the short-hand turtle commands like (forward) as well.

zoeyoung12, 2017-07-17:
I had the same question. Thanks for the response