google/coding-with-chrome #113: Support for Wowwee MiP robot?

optionaltoast, 2017-06-13:
Hi

Is it possible to add support for the Wowwee MiP robot to the app? Wowwee have made the SDK available here https://github.com/WowWeeLabs
There are a number of projects which use this, the Cylon.js library works but unfortunately the Scratch2MiP project doesn't.  

Thanks

MarkusBordihn, 2017-06-13:
Unfortunately Wowwee MiP is using an Bluetooth LE connection which is currently not supported by Chrome on all Platforms out of the books.

optionaltoast, 2017-06-13:
Thanks for the prompt reply. I may be clutching at straws but should I read anything into the phrase "currently not supported" will this feature be added at a later date?

MarkusBordihn, 2017-07-24:
As soon we have Bluetooth LE support for all platforms we would be able to add such integration.
Currently there is unfortunately no real solution which worked for all platforms like Chrome OS, Windows, Mac and Linux.

closed 2024-02-14 by MarkusBordihn