google/coding-with-chrome #116: Yes/No buttons are bad(tm)

MartenvanWezel, 2017-06-13:
The current format of the dialogs that we have is:

"Changes have not been saved. Exit?"
[no] [yes]

Current best practice for UI is:
"Changes have not been saved. Exit?"

[cancel] [exit]

We should review all dialogs and update.

raghukul01, 2017-06-24:
Does this issue only require the change of the [yes] [no], or some other UX improvements? Please clarify, I am willing to do this task.

MartenvanWezel, 2017-06-25:
Hi raghukul01,

Thanks for your help offer. We would be happy to accept! 

We usually split up tasks in smaller chunks that are small and easy to describe, so to fix this issue it would only be the buttons.

This of course does not mean more work can't be done on the UI, but the review I sat in was fairly small, so these were the only active points. If you see any other things that clearly could be improved then of course you can offer more pull requests ;)

raghukul01, 2017-06-26:
Thank you Sir for your response, I would try to finish as early as possible

MartenvanWezel, 2017-10-05:
Hi raghukul - shall we assign this back to someone else, or will you finish this?

TioBorracho, 2018-03-09:
taking this issue

TioBorracho, 2018-03-09:
I didn' t want to change the parameters in the render function, so I used the opt_values to send the action string to the template
should I pull request?
https://github.com/TioBorracho/coding-with-chrome/commit/2f6232c8ffd02fd18c61fc83e2af3fa2900b3a2e

MartenvanWezel, 2018-03-09:
Yes please =)

MarkusBordihn, 2018-03-09:
Thanks @TioBorracho

closed 2018-03-09 by MarkusBordihn