google/coding-with-chrome #129: Save functionality is inconsistent, multiple issues

rowbottomn, 2017-12-08:
Sometimes when saving a project, no feedback is given that saving was successful, then when you try to close, you get the warning you will lose work.  When you reopen, sometimes you have lost work and other times its ok.

Will try to get more feedback on this as we are using this in a class.

Currently this issues is for sure with blockly, not sure about javascript.

rowbottomn, 2017-12-11:
Yesterday...I saved a project offline then signed in using the icon.  Saved on my drive.  No error messages or warnings.  Went back in and project was not in my Google drive.  I will check and see if I can reproduce error. 

I will have my class save in two locations and then check for Google drive existence.

rowbottomn, 2017-12-11:
It did not save to my hard drive either :(

rowbottomn, 2017-12-11:
This image shows the file in my google drive
[image]
This image shows it warning me file has not been saved
[image]

Note that it occurs to my class that the file name has not been update to reflect the name of the saved file.  This could be the source of the error.

rowbottomn, 2017-12-12:
This is an issue that has come up quite a few times when saving to google drive, Xhr request error!
Logging out and back in seems to fix.
[image]

rowbottomn, 2017-12-13:
Additional issue....left project open all day and upon returning, can no longer save on google drive.  Tried signing out and back in to google drive and selecting save in google drive is unresponsive.  Also saving as new file is unresponsive.

MartenvanWezel, 2017-12-13:
Thanks Nathan. 

1/ Can you confirm that the issue also happens with local file systems now, or is this primarily a Google Drive issue?

2/ Just to doublecheck- you are using the latest CwC? (which version?)

rowbottomn, 2017-12-13:
Marten,

Thanks for responding.
1./  Today we has multiple occurrences of not being able to save like I reported yesterday. A little frustrating for students to have a class worth of work but not be able to save with either method.  I had them log out and then sign back in to chrome  using the in app interface.  It helps when they receive the Xhr request error but not with this.

It does not give really any feedback at all.  The selection to save is just ignored. I believe the rest of the menu is active like to start a new project.  It is weird.

2./ I am using 4.5.16 and the rest of the class is as well..

Thanks again for your help, I think this project has amazing potential.  Any suggestions for how we can help you with more informative feedback?

Nathan

MartenvanWezel, 2017-12-13:
Not sure - I've looped in the dev responsible for this feature, hopefully he can give more directions. Sorry for the frustration- we'll try to address it ASAP.

rowbottomn, 2017-12-20:
Not being able to select saving options in the menu is probably the largest issue with using this application.   Students get fatigued with having to reproduce work over and over because they cannot save.  We are saving the javascript as gists between day to day but then we are losing the ability to use blockly as a method to rapidly prototype.  Any improvements to this would, in my mind, be a critical issue.   On any day 30% of the class cannot save with out any exaggeration.

MartenvanWezel, 2017-12-22:
Quick update: 

1/ To be clear, is this a Google Drive issue, or does this happen with normal filing system too?
2/ What hardware are you using? ChromeOS? Windows? Mac? Version of OS would be helpful?

Basically we have not been able to reproduce this yet.. Does this happen seemingly 'randomly' or have you noticed some pattern? Poor network connectivity? Only after midnight and full moon? Etc.

MartenvanWezel, 2018-01-09:
Hi all, we've updated CWC to 4.6.25 - can you check if this helps anything?

closed 2018-03-19 by MarkusBordihn

rowbottomn, 2018-03-19:
Sorry...that class is over.  I will check and get back to you anyway.

-- 
Nathan Rowbottom,
Teacher, Craig Kielburger S.S.
Physics, Science and Computer Science

-- 
This e-mail and any attachments are the property of the Halton District 
School Board and are  intended only for the use of the addressee(s) and may 
contain information that is privileged and/or confidential and/or protected 
under the Education Act, the Municipal Freedom of Information and 
Protection of Privacy Act and/or the Personal Health Information Protection 
Act.  Unauthorized review, distribution, copying or disclosure is strictly 
prohibited. If you are not the intended recipient please notify the sender, 
delete this message and do not print, copy, distribute or disclose it 
further.