google/coding-with-chrome #130: Additional functionality: ability to move code between projects.

rowbottomn, 2017-12-11:
Love this project and want to support it.  Just posting suggestions that would enhance usage in educational context.

Having students prototype aspects in Blockly and then merge into a larger project is made difficult by the inability to have multiple instances open of coding with Chrome. 

Only way is to copy and paste markup language.  Scratch has a backpack functionality for blocks to go between projects.  This would be very useful.  Also the ability to copy JavaScript between projects in app would be nice.  I understand the burden of running two projects would be taxing but if perhaps they were disable from running while in an certain mode.

MartenvanWezel, 2017-12-13:
Suggestions are very welcome - thanks.

Multiple instances, interesting idea. I don't think we can run a Chrome App twice in parallel, but we could consider 'open new window'. Of course we need to think about making things complicated through adding too many infrequently-used features. 

Can you tell me a use case of how you would imagine multiple students create (collaborate?) on a larger project? 

Not sure what you mean by 'blocks to go between projects' you mean a 'block of blocks', e.g. cut-and-paste an entire routine? The backpack as I understand it really isn't that much more than a glorified cut/paste buffer, isn't it?

rowbottomn, 2017-12-13:
A glorified paste bucket is most likely what I am talking about.  

A use case is say perhaps students are learning to draw air hockey paddles using the draw commands.  They save the blocks in that project.  

We then move on to a more complicated game example to spin off into an air hockey game (the bouncing cubes) which features no draw commands.  I understand we would still need a physics sprite that we would draw over top of in order to handle the puck interactions.

Perhaps this would be a good time to ask why the different use cases have such a variance in the availability of blocks available?  I understand that the beginner mode is meant for users that might need a heavily simplified interface with the block to avoid from being overwhelmed, but it would be nice to be able to  break down a larger project in to simpler programs and then be able to assemble them together.

Here is a Scratch example of the backpack:  I can fill it with the prototypes of common problems that need to be solved and then deploy them for new projects.  Sorry in advance for pasting in a large image.
Its at the bottom.
[image]

closed 2024-02-14 by MarkusBordihn