google/coding-with-chrome #133: Translation to Hindi

SetCodesToFire, 2018-01-21:
Hello I would like to make a Hindi translation to make coding-with-chrome more accessible.

Can i contribute in this project? I have signed the CLA. My understanding with a previous issue #88 shows me that i need to create a local/[language-initials]/translation.js

What name should I pick for Hindi?

MartenvanWezel, 2018-01-21:
Yes!

Thank you SetCodesToFire, it is alway good to have someone who understands the code and language to do this. The international language code for hindi is "hi".

Please go ahead and create a pull request when you are ready. Indeed it is

locales/hi/translation.js

-M

SetCodesToFire, 2018-01-21:
Welcome. I will create a PR after some time.

MarkusBordihn, 2018-01-21:
Thanks a lot for offering your support on this, please create a new file with the name:
local/hin/translation.js for the Hindi translation.

You could use local/eng/translation.js as reference or just start translating the main parts.

I will update the [I18n instruction](https://github.com/google/coding-with-chrome/blob/master/doc/I18N.md) with the workflow for testing such changes.

closed 2018-01-23 by SetCodesToFire

MartenvanWezel, 2018-01-23:
Thank you SetCodesToFire - we intend to push a new version containing your code in a few days.

SetCodesToFire, 2018-01-23:
Welcome @MartenvanWezel ,

I want to know where to start looking for contributing into this project more. I see there are many commits so i scrolled way back to the very first commit of the project and found myself lost.

MartenvanWezel, 2018-01-23:
Hi - please feel free to send me a mail at codingwithchrome@google.com  - we can discuss in detail.

(but of course you can just start taking issues, but maybe you feel like a bigger project, we have tons of ideas and not enough time ourselves =) )

Cheers,

-Marten

SetCodesToFire, 2018-01-23:
I sent you a mail... Subject mentioned is

setCodesToFire to Marten van Wezel 

:)