google/coding-with-chrome #139: Import images for EV3 coding

MrSmith1988, 2018-02-22:
Would be awesome if users could import .png files with colours into the simulation. The simulated robot would only need to detect the colour of the pixel (assumedly in hex) it is currently touching with the top of the triangle on the screen, and return that into the colour sensor.
This would be useful for testing line following programs in simulation for schools without too many resources.
 I have no hope of doing this myself...

MarkusBordihn, 2018-03-30:
@MartenvanWezel 
I agree that it would be nice to have an interactive Robot simulation. Would be great if you could place this in the roadmap like "Interactive and universal Robot simulation which supports, actors and sensors for all kind of robots."

MartenvanWezel, 2018-04-06:
Markus: can we not use the Open Roberta simulator?

MrSmith1988, 2018-04-06:
From the looks if it, Open Roberta would be an amazing tool to use to simulate robots with CWC. I would say to ditch CWC and just go with Roberta, but CWC still seems to have some benefits like:
- automatic integration with Google Drive and Classroom
- simpler text editor for EV3 robots instead of roberta's C++ text editor which would be extremely intimidating for beginner programmers
- Ability to plug in real robots without having to flash with SD card.

So it would be awesome to combine the two together. Roberta is open source, right? Anyone able to grab the simulation system from Roberta and shove it into CWC?

closed 2024-02-14 by MarkusBordihn