google/coding-with-chrome #140: Student code fails to execute sometimes

MartenvanWezel, 2018-02-23:
**Note: typically restarting Chrome helps**

Scattered reports are coming in that sometimes code is not being parsed, e.g. the Python Example is opened but the turtle doesn't show up.
[image]

Given that typically restarting Chrome helps, the suspicion is that this is a Chrome issue not a _Coding with_ Chrome issue, but still, please **Add to this Issue** if you experience it too.

Quick-fix:
Mac/Win Chrome: You might have to actually fully restart Chrome
ChromeOS: Right-clicking in the main CWC window, and selecting 'Simulate Browser Restart' should help.

MarkusBordihn, 2018-02-23:
Please check the console over the Chrome Web Developer tools to get additional informations about at the time the issue is happen.

If this is mostly happen for the Python Integration it could be related to the built-in webserver.

MartenvanWezel, 2018-02-23:
It's not Python related, we had the same issue with Samuel's cloud vision demo. Will try to get dev tools output.

MartenvanWezel, 2018-03-02:
Also qq: given that this is a chrome app, how do I switch to the console?
Is there some clever way? Or can I just click on Chrome, dev tools etc? I tried that one and I don't see anything helpful...

MarkusBordihn, 2018-03-02:
You could bring up the "dev tools" if the developer mode is enabled or you go to the Extension settings and using "Inspect views" for "html/editor.html" instead.

closed 2018-03-02 by MarkusBordihn