google/coding-with-chrome #165: Create a Notification Queue

MartenvanWezel, 2018-05-23:
Many sites have a notification queue, facebook and G+ are fine examples, where the icon is greyed, and shows a red circle with a number if new important events have happened.

@MarkusBordihn @efuquen @averypescado @matt-diehl would it make sense we create some type of notification queue, where 'important CWC events' show up? 

(WBench) - "New tutorial projects downloaded"
(CWC Core) - "Coding with Chrome has updated"
(Classroom) - "New assignments for you from Mr. Rogers"

etc..

Facebook example attached.

[image]

MarkusBordihn, 2018-05-23:
Honestly I don't know if this really make sense in the general UI.
It could be maybe added in the start screen, but it makes no sense in the core UI because its not really an re-usable UI part or Icon.

MartenvanWezel, 2018-05-23:
Not sure I understand what you mean by re-usable UI part?

The main flow I imagine would be:

- New projects downloaded from WB -> notify with an icon -> click -> get
taken to projects view
- CWC version updated -> notify -> click -> show 'changelog'
- New assignments from classroom -> notify -> click -> show new assignments

I would want this primarily in the main UI and not in the 'select your
skill or mode' screens.

MarkusBordihn, 2018-05-23:
Re-usable means that it will be used more than once per user session.
In this case such notification will be only used once  and that it will not be used for a long time.

So I'm fine with this in some screen, but not in the main UI like titlebar, toolbar, sidebar or anything like this.

MartenvanWezel, 2018-05-23:
But like I said, I think it'd be reused all the time. If you're in an
active class you would conceivably get quite a few notifications of new
assignments, or perhaps a notification that your bluetooth device
disconnected etc etc. No?

averypescado, 2018-05-25:
> Got it. So you want to move all existing notification to an single
> notification system / icon. SGTM.

My thought is that if the main use case is assignments, assignments should
simply be broken out in its own page, with the plus one there when one is
added.

Since the relative importance of a new assignment for a student is much
larger than that of an update we did, I don't think they should be grouped
together.

Also, in most other academic portals I have used there is a specific page
for assignment, which I found very useful from an organizational
perspective.

MartenvanWezel, 2018-05-25:
Thanks Avery, agreed to break it out in a separate page, but we want to
alert the user that they should 'check out the assignments page', no?

averypescado, 2018-05-30:
> issuecomment-391817437
> AkAIbVIYCUGW4DVX3B6ftgZBlHBopMZsks5t1v4xgaJpZM4UJm7p

Yes, I think that page should have its own notifications

closed 2024-02-14 by MarkusBordihn