google/coding-with-chrome #170: CwC icon bouncing on MacOS

jmedward, 2018-06-14:
I've noticed that the CWC icon keeps bouncing on the my mac dock when I minimize it. Typically the bouncing will continue for 30 seconds or so and eventually stop. However, sometimes it randomly starts bouncing again. When i click on the icon, it opens the CWC window and the icon stops bouncing.

MarkusBordihn, 2018-06-14:
Thanks for the report, seems related to the way Chrome Apps are handling this.
Will be fixed in the next version.

MarkusBordihn, 2018-06-15:
I removed all additional "attention" calls from the source code, but the icon is still bouncing in some rare instances.

This seems to be the expected behavior for Chrome Apps on Mac OSX, so there is not a lot which could done from our side.

closed 2018-06-15 by MarkusBordihn