google/coding-with-chrome #171: npm run sync

SIMAKK, 2018-06-14:
I am having issues with npm run sync on windows 10 machine

Failed at the coding-with-chrome@5.6.12 sync script

I have tried verifying cache, reinstalling npm but still no joy!

Here's my log 

[2018-06-14T07_08_57_662Z-debug.log](https://github.com/google/coding-with-chrome/files/2101161/2018-06-14T07_08_57_662Z-debug.log)

MarkusBordihn, 2018-06-14:
Very strange, it's not even executing the "echo" command.
Please execute the commands manual and let us know which command fail.
```
echo Syncing. Please wait ...
git pull
git submodule init
git submodule update
npm run update
```

Make sure you are using the PowerShell on Windows 10.

SIMAKK, 2018-06-15:
npm run update seems to be the culprit. Check my log now...

[2018-06-15T12_34_31_524Z-debug.log](https://github.com/google/coding-with-chrome/files/2106051/2018-06-15T12_34_31_524Z-debug.log)

MarkusBordihn, 2018-06-15:
Please run `npm install` first before executing any other `npm run ...` command.

closed 2018-06-15 by MarkusBordihn