google/coding-with-chrome #174: Forward and Back Navigation

rachwein, 2018-06-20:
[Add Back & Forward Arrows.pdf](https://github.com/google/coding-with-chrome/files/2120840/Add.Back.Forward.Arrows.pdf)

MarkusBordihn, 2018-06-21:
Thanks for the UX research, unfortunately the arrow idea is not doable in this case.
The tab bar will already use arrows if the content is to small.
This means by adding these additional arrows it will create confusions when both arrows showing up.

closed 2018-06-21 by MarkusBordihn

MartenvanWezel, 2018-06-21:
Hey Markus, I think we could head off this situation by just hiding the other arrow?
It seems helpful to me to always have a back arrow "somewhere"?

MartenvanWezel, 2018-06-21:
The entire UX team agreed earlier that it was hard to find the way back to 'home' so we should do this somehow..

MarkusBordihn, 2018-06-21:
Hey Marten, please checked my latest changes in https://github.com/google/coding-with-chrome/commit/bc0bec4afb3af2900f0fe678f18becd6d70c7a43 which should solve most of the suggested changes.
But as I mentioned with our current responsive design its not possible to implement these arrows.
See https://material-components.github.io/material-components-web-catalog/#/component/tabs under Scrolling Tabs.

rachwein, 2018-06-21:
> Hey Marten, please checked my latest changes in bc0bec4
> web-catalog/#/component/tabs under Scrolling Tabs.

Hi all,

Another idea that may be simpler and a quick fix is to keep the flow as is
and change the wordage of "New Project" (which takes you to the home screen
under options) to "Switch to Advanced Mode" or "Back."

-Rachel