google/coding-with-chrome #188: Open Tasks (refactoring)

MarkusBordihn, 2018-09-18:
List of open Tasks

- [ ] src/ui/gdrive/gdrive.js move multipart encoder (saveFile) to separate tool.

- [ ] src/frameworks/simple/draw.js move drawInstruction to an separate file.

- [ ] src/file_format/file_format.js move deprecated handler to separate file.

- [ ] General replacement for `let x = ... if (x) { do something with x}`

- [ ] Using this.events_.listen feature consistently.

- [ ] Replace src/utils/... with coding-with-chrome-libraries alternatives.

- [ ] Move preview key listerner for CTRL+Enter and CMD-Enter src/ui/editor/editor.js to src/ui/preview/...

mchav, 2018-11-01:
These seem to be a good place to start contributing. Is there a doc detailing what each entails? The 4th one is very vague to me.

MarkusBordihn, 2018-11-02:
The 4th one is for all cases like:
```
  let userConfigInstance = this.helper.getInstance('userConfig');
  if (userConfigInstance) {
    this.enabled = userConfigInstance.get(cwc.userConfigType.GENERAL,
        cwc.userConfigName.DEBUG_MODE) || false;
  }
```

I would like to have a clever way to avoid such constructs and maybe replacing the singleton *this.helper.getInstance* function with some better solution or global library instead.

A good start point is https://codeclimate.com/github/google/coding-with-chrome/issues?category=complexity&engine_name%5B%5D=structure&engine_name%5B%5D=duplication which gives you an good indicator which parts could be optimized or so.

closed 2024-02-14 by MarkusBordihn