google/coding-with-chrome #235: shepherd.js usage

rwwagner90, 2019-06-09:
👋 Hello! I see you are using shepherd.js in this project. We are so pleased to see that! We would love to showcase your usage in the Shepherd docs, and we would love to hear how things have been going for you using it. What has worked well, do you have any feature requests or improvements, etc. Please let us know!

MarkusBordihn, 2019-06-10:
Hi Robert, thanks for checking. Your library really simplifies a lot for our project.
We are using shepherd for our tutorial engine, beginners tour and some in app help / guidance and would be happy to be mentioned under showcases in your docs.

Currently we we have no features requests or improvements recommendations.

Thanks a lot for the great work and shepherd is already linked inside the app and under https://github.com/google/coding-with-chrome/blob/master/NOTICE.md from our side.

rwwagner90, 2019-06-10:
Awesome, thanks @MarkusBordihn! Would you perhaps want to open a PR adding a brief description of your project in this section? https://github.com/shipshapecode/shepherd/blob/master/index.md#websites-and-apps

rwwagner90, 2019-06-21:
@MarkusBordihn I would be happy to add a blurb as well, if you wanted to provide something small here instead or opening a PR? Just let me know!

closed 2024-02-14 by MarkusBordihn