google/coding-with-chrome #260: README 404

DamieFC, 2020-12-14:
I was looking at the README.md file at the start, and there is a sentence that says "If you are looking for the Chrome App version, please go to the chrome-app-v6.x repository instead", and there is a link at the "chrome-app-v6.x repository" bit that goes to a site at "github.com/google/coding-with-chrome/blob/main/1". Weeeeeeelll, I went to click it, and I got the Github 404 page not found page. Sorry if this isn't really important, something was nagging me to put this here 😞

MarkusBordihn, 2020-12-17:
Thanks a lot for the report. I fixed the typing error in ce785de6514aaf8512eda89111c5da70097ff5dc.

closed 2020-12-17 by MarkusBordihn

DamieFC, 2020-12-18:
Thank you for fixing that for me!