google/coding-with-chrome #278: Is there a nested git repository inside this project?.

SampathKumar27896, 2023-04-15:
I am trying to run the project in my localhost. 
First I executed
`npm install`
Then I faced the issue
`Module not found: Error: Can't resolve '../../../../third_party/phaser-ce/build/phaser.min.js`
I think the project depends on the package **pahser-ce**  which seems like a nested git folder on the above mentioned path.

MarkusBordihn, 2023-04-16:
Thanks for your feedback. It seems you cloned the repository with the `--recursive` flag so the required git sub-modules are missing.

You can try to run `git submodule update --init --recursive` to get them or cloning the repository with the `--recursive` flag instead.

closed 2023-04-16 by MarkusBordihn

SampathKumar27896, 2023-04-18:
> Thanks for your feedback. It seems you cloned the repository with the `--recursive` flag so the required git sub-modules are missing.
> 
> You can try to run `git submodule update --init --recursive` to get them or cloning the repository with the `--recursive` flag instead.

This is something I have never tried. I take down this one.