google/coding-with-chrome #295: Tabs on interface

alpaslan-exe, 2024-04-26:
Note: This is a UX issue rather than a direct code bug. 

I quickly noticed that upon launching an instance of the editor, the menu to the left where blocks are chosen from is not uniform in its function. When clicking certain categories of blocks, the current blocks shift slightly, whilst, with bigger tabs, the category opened will simply overlay over the current blocks chosen. There are two solutions that I can see to this problem: either make all menus overlay the code without a shift or simply make all menus regardless of size shift the blocks further to the right. If anybody is able to suggest an alternative solution I would love to hear your ideas. This is likely quite an easy fix in terms of the actual menu code. For reference, I will include a screenshot of the components marked to highlight the components I've discussed here. 
[image]

alpaslan-exe, 2024-04-26:
Note that this issue was observed on Chrome 124.0.6367.62 for macOS ARM

MarkusBordihn, 2024-04-27:
Thanks a lot for the feedback. Unfortunately this is by design from the used "Blockly" framework, see:
https://google.github.io/blockly-samples/examples/toolbox-demo/

Feel free to open a corresponding feature report on https://github.com/google/blockly instead.

closed 2024-04-27 by MarkusBordihn